generalized-intelligence/GAAS #17: Unsupported FCU when started offboard script

AdityaGawali, 2019-03-26:
After following the tutorial 2 and launching the 'sfm.launch'
 file and in new terminal starting 'px4_mavros_run.py'
 I get this error `[ERROR] [1553594972.501983904, 162.996000000]: MODE: Unsupported FCU` in the terminal where i launched the 'sfm.launch' file.

MasterPa, 2019-03-26:
Got it! Thank you for feedback. We will discuss this problem as soon as possible:)

Doodleshr, 2019-03-27:
I tried on a brand new PC but I cannot seem to reproduce the same error.
Does `roslaunch px4 mavros_posix_sitl.launch` work? Can you give it a try? If it doesn't the problem would be unrelated to GAAS and I would recommend you to rebuild px4.

AdityaGawali, 2019-03-27:
I did the roslaunch command you mentioned the gazebo world with a bungalow (village.world) opens successfully but when I run px4_mavros_run.py it throws unsupported FCU error.

The error does not come when I do make_px4_sitl_default gazebo

AdityaGawali, 2019-03-29:
I got the real issue causing it but don't know how to solve it.
After launching `sfm.launch` and doing `rostopic echo /mavros/state` , I get response as `connected: False`. Basically Mavros isn't getting connected to my simulator

closed 2019-03-29 by AdityaGawali