generalized-intelligence/GAAS #19: When starting launch files via roslaunch "roslaunch px4 sfm.launch"- ERROR [px4] Error opening startup file, does not exist: etc/init.d/rcS

lulu402, 2019-03-29:
**Followiing is the log:**

lulu@lulu-Vostro-3668:~$ roslaunch px4 sfm.launch
... logging to /home/lulu/.ros/log/3014fc68-5133-11e9-b6c3-3cf862c629e7/roslaunch-lulu-Vostro-3668-22089.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lulu-Vostro-3668:45265/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
[... 117 lines cut ...]
[ INFO] [1553823913.508733130]: Plugin command initialized
[ INFO] [1553823913.508822193]: Plugin ftp loaded
[mavros-5] killing on exit
[ERROR] [1553823913.512777707]: [registerService] Failed to contact master at [localhost:11311].  Retrying...
[vehicle_spawn_lulu_Vostro_3668_22089_7992864178812158555-4] killing on exit
[gazebo_gui-3] killing on exit
[sitl-1] killing on exit
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/gazebo_ros/spawn_model", line 21, in <module>
    import rospy, sys, os, time
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/__init__.py", line 49, in <module>
    from .client import spin, myargv, init_node, \
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 54, in <module>
[gazebo-2] killing on exit
    import rospy.core
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/core.py", line 66, in <module>
    import rosgraph.roslogging
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosgraph/roslogging.py", line 41, in <module>
    import logging.config
  File "/usr/lib/python2.7/logging/config.py", line 489, in <module>
    class DictConfigurator(BaseConfigurator):
KeyboardInterrupt
shutting down processing monitor...
... shutting down processing monitor complete
done

Doodleshr, 2019-03-29:
`INFO  [px4] Creating symlink /home/lulu/src/Firmware -> /home/lulu/.ros/etc
ERROR [px4] Error opening startup file, does not exist: etc/init.d/rcS`

That's odd.
Check if ~/.ros/etc/init.d/rcS exists, otherwise you have to manually copy and paste it to corresponding folder.

Doodleshr, 2019-03-31:
Close as stale, reopen if needed.

closed 2019-03-31 by Doodleshr