generalized-intelligence/GAAS #21: Is there any solution to control multi-uavs with the GAAS

myname1234, 2019-04-14:
I want to know ,how can i control multi-uavs with the GAAS, i'm new to  python and mavros.

MasterPa, 2019-04-15:
So what kind of multi-UAVs you mean? Or what do you want to do with multi-UAVs?

myname1234, 2019-04-15:
> So what kind of multi-UAVs you mean? Or what do you want to do with multi-UAVs?

I mean control multi-UAVs to move toghter on the same time, like formation. I know the px4 Firmware launch has multi_uav_stil.launch，but i can't use the commander.py to control it directly. How do i change the py file.

MasterPa, 2019-04-15:
GAAS didn't support control multi-UAVs yet, but we will support it in the future(>6 months maybe). You could leave your requirements here:)

Doodle1106, 2019-04-18:
closed as stale, reopen if needed.

closed 2019-04-18 by Doodle1106