generalized-intelligence/GAAS #22: How to use APM 2.8 or Naze32 flight stacks instead of pix4?

atmadeep, 2019-04-17:
how can we import flight stacks other than pix4? I have two drones each built using naze32 and APM 2.8 . i don't even know if any of the above mentioned are supported or not.

cyanine-gi, 2019-04-17:
Currently we do not support apm officially. We will support apm in the future. You can come up a pr if you wish, just deal with the mavlink msgs, which will be helpful.

atmadeep, 2019-04-17:
Thanks for the reply, I'll get to work on it right away. Can you please provide me more info on this? As in some examples which I can refer to gain more understanding.

bys1123, 2019-04-20:
APM 2.8 is a abonded hardware, but Ardupilot w/ Pixhawk is possible.

atmadeep, 2019-04-20:
I know it's depecrated, but can you please tell me any way just for this
one project? That's why it came in cheap. I swear to almighty that I'll
abandon it too.
On a second thought, How about an APM 3.1?
Waiting to hear from you,
Atmadeep Arya.

AlexisTM, 2019-04-24:
I guess you mean Ardupilot. Currently Gaas is using OFFBOARD mode which does not exist for Ardupilot.

atmadeep, 2019-04-24:
Is there no way I can use this with an APM 2.8 or even APM 3.1? These 
are the only boards I have available.
Can I at least run simulation on the images collected using my drones? 
If yes, can you please tell me how?

AlexisTM, 2019-04-24:
If it is the hardware, yes you can put the PX4 Firmware onto your device. I guess the APM target is gone now.

Doodle1106, 2019-05-07:
thanks @AlexisTM  for helping.

Close it for now as it is stale. Reopen if needed.

auturgy, 2019-05-07:
There’s confusion here between the old 8 bit APM flight controller and the ArduPilot firmware, which runs on more than 40 different hardware targets. The firmware is very much alive and well, so leave this open

atmadeep, 2019-05-28:
Can I use pixhawk 2.4.8 for this project? I can buy one. Will it be supported in the long term in terms of px4 firmware development?

MasterPa, 2019-05-28:
Yes, you could use pixhawk 2.4.8.

AlexisTM, 2019-05-28:
You can use any of the following hardware: https://docs.px4.io/en/flight_controller/pixhawk_series.html

The contestants being:
- Pixhawk 4
- Pixhack v5
- Pixhawk 2 (branded Cube)

It is important to know that the number is **unrelated** to a version number. It is only marketing and these come from different suppliers. The different suppliers are using the FMUv{1,2,3,4,5,6} specifications that gives the general specs and pinout of each board.
Pixhawk 4 and Pixhack v5 are both FMUv5. FMUv6 is under specs development and will expose Ethernet and CAN buses with industrial connectics.

From what I heard, Pixhackv5 has a very good finish. I am personally using a lot of cubes (Pixhawk 2).

[image]

closed 2021-04-27 by cyanine-gi