generalized-intelligence/GAAS #29: Warning when compiling YGZ-slam

jiahua99, 2019-05-09:
Please provide any relevant information about your setup.
* Tutorial No.:Tutorial 3
* Mavros: 0.18.4
* ROS: Kinetic
* Gazebo:7
* Ubuntu: 16.04
* System Platform: X86
*OpenCV :3.3.1

I'm trying to compile the YGZ-slam by "sh generate,sh" and i ran into the warning as shown below. Despite that, i'm still able to compile it successfully. May i know if this warning will affect the performance of the slam.

Warning log:
Could NOT find ensenso (missing:  ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** io features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing:  DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** io features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing:  _DSSDK_LIBRARIES) 
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
-- Could NOT find ensenso (missing:  ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** visualization features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing:  DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** visualization features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing:  _DSSDK_LIBRARIES) 
** WARNING ** visualization features related to dssdk will be disabled
-- Could NOT find RSSDK (missing:  _RSSDK_LIBRARIES) 
** WARNING ** visualization features related to rssdk will be disabled

MasterPa, 2019-05-10:
Thanks for the report, we will review it soon:)

Doodle1106, 2019-05-10:
No, the problem is PCL related and we are currently freezing PCL related functions in SLAM, so these warnings will not impact slam performance.

Are you using Visual Studio? PCL tend to have dependency issues with Visual Studio.

closed 2019-05-10 by Doodle1106