generalized-intelligence/GAAS #30: bad results report

guoxianzhen, 2019-05-10:
# Issue Template

## Context
Please provide any relevant information about your setup.
* Mavros: [0.18.4]
* ROS: [ Kinetic]
* Gazebo: [ 7]
* Ubuntu: [16.04]
* System Platform: [X86]

## Expected Behavior
I hope the copter could perform better according to the command.
## Current Behavior
Takeoff:
Sometimes when the copter takes off, it will fly higher than the expected location and then hovering around the destination. Sometimes, the copter may flies randomly after taking off.
Land:
During the landing process, the copter will keep fly forward. I hope the copter could land just right below or close to the point where it covering.
Flying mission
The copter has not succeeded flying in a square way. The slam will lose when the copter turning 90 degrees and then the copter will fly randomly. 
In addition, according to the test tutorial, I have installed all the dependencies including Pangolin EIgen3 g2o PLC and Opencv.
I just want to ask it there results are normal or if I have missed some steps.
## Screenshots
These are some screenshots of the trajectories when copters takes off.
[image]

[image]

[image]
I also simplifies the mission in square.py. I command the copter to fly to the right for 3 meters then turn around for 90 degrees, land finally. This is the results
[image]

## Failure Logs
**part of the console of slam :*********** Tracking frame 2150 **********
I0510 16:05:06.864307 26386 TrackerLK.cpp:110] Tracking frame 1
I0510 16:05:06.879122 26389 BackendSlidingWindowG2O.cpp:133] new good points: 0 in total immature points: 0
I0510 16:05:06.879212 26389 BackendSlidingWindowG2O.cpp:156] Local BA cost time: 9.9e-08
I0510 16:05:06.879249 26389 BackendSlidingWindowG2O.cpp:182] Deleting KF 0
I0510 16:05:06.879266 26389 BackendSlidingWindowG2O.cpp:193] Erase KF in mpKFs
I0510 16:05:06.879516 26389 BackendSlidingWindowG2O.cpp:199] Done
I0510 16:05:06.879671 26389 BackendSlidingWindowG2O.cpp:427] Clean map point done, total bad points 1
I0510 16:05:06.879696 26389 BackendSlidingWindowG2O.cpp:165] Backend KF: 10, map points: 132
I0510 16:05:06.879714 26389 BackendSlidingWindowG2O.cpp:38] Process new KF done.
I0510 16:05:06.902025 26386 Tracker.cpp:1568] Quit iteration in: 10
I0510 16:05:06.904637 26386 Tracker.cpp:1586] Inlier/total=129/129
I0510 16:05:06.904803 26386 Tracker.cpp:1602] Estimated TCW = 
0.0767104 00.983975 00.160963 0014.6661
-0.983889 0.0485575 00.172059 004.61203
00.161486 -0.171568 00.971847 0010.8394
000000000 000000000 000000000 000000001
I0510 16:05:06.932163 26386 ORBExtractor.cpp:470] Calling fast single level
I0510 16:05:06.935111 26386 TrackerLK.cpp:541] Detect new feature cost time: 0.00319594
I0510 16:05:06.938956 26386 TrackerLK.cpp:613] new stereo: 0, new Mono: 0, update immature: 0, total features: 129
I0510 16:05:06.939127 26386 TrackerLK.cpp:618] Create map point cost time: 0.00722153
I0510 16:05:06.939270 26386 Tracker.cpp:506] Insert keyframe done.
I0510 16:05:06.939328 26386 TrackerLK.cpp:235] Insert KF cost time: 0.0074259
I0510 16:05:06.939479 26386 TrackerLK.cpp:114] Tracking frame 2
I0510 16:05:06.939540 26386 TrackerLK.cpp:148] Tracking frame 3
I0510 16:05:06.939602 26386 TrackerLK.cpp:153] Tracker position = 1.68634,-12.7308,-13.6894,
vision pose: -14.9427, -10.4848, -4.37022
EstimatedPose : -14.9427, -10.4848, -4.37022
I0510 16:05:06.941656 26389 BackendSlidingWindowG2O.cpp:36] Process new KF
QObject::~QObject: Timers cannot be stopped from another thread

Console of px4 Firmware
... logging to /home/guo/.ros/log/bcb47986-72f9-11e9-9dbb-b0c0902a3964/roslaunch-guo-Lenovo-TianYi-100-14IBD-25090.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/loader.py:409: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
[... 346 lines cut ...]
[ INFO] [1557475260.378267336, 1.262000000]: IMU: High resolution IMU detected!
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
[ INFO] [1557475261.238106135, 2.064000000]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1557475261.248689658, 2.072000000]: IMU: High resolution IMU detected!
[ INFO] [1557475261.361614057, 2.176000000]: IMU: Attitude quaternion IMU detected!
[ INFO] [1557475262.705694238, 3.064000000]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1557475262.705787152, 3.064000000]: VER: 1.1: Flight software:     010800ff (96443b3cf3e3adae)
[ INFO] [1557475262.705841238, 3.064000000]: VER: 1.1: Middleware software: 010800ff (96443b3cf3e3adae)
[ INFO] [1557475262.705885628, 3.064000000]: VER: 1.1: OS software:         040f00ff (0000000000000000)
[ INFO] [1557475262.705926492, 3.064000000]: VER: 1.1: Board hardware:      00000001
[ INFO] [1557475262.705969890, 3.064000000]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1557475262.706015072, 3.064000000]: VER: 1.1: UID:                 0000000100000002
[ INFO] [1557475279.586635608, 12.064000000]: HP: requesting home position
[ INFO] [1557475290.977850830, 17.064000000]: WP: mission received
[ INFO] [1557475306.203129558, 22.064000000]: HP: requesting home position
INFO  [ecl/EKF] EKF commencing external vision position fusion
[ WARN] [1557475321.681447912, 26.810000000]: CMD: Unexpected command 246, result 0
[ INFO] [1557475336.404410881, 32.064000000]: HP: requesting home position
WARN  [commander_tests] Failsafe enabled: no RC
[ WARN] [1557475358.762256937, 39.422000000]: CMD: Unexpected command 176, result 1
[ INFO] [1557475358.778073944, 39.426000000]: WP: reached #0
[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/iris_stereo_gray_no_gps/gazebo/command/motor_speed, deleting message. This warning is printed only once.
[ INFO] [1557475358.839747618, 39.450000000]: WP: reached #0
[ INFO] [1557475359.078308181, 39.530000000]: WP: reached #0

ninawrong, 2019-05-12:
Thank you for submitting the issue :)
We will get back to you on Monday.

Doodle1106, 2019-05-14:
Poor performance is related to optimizePoseFaster()， which replaced optimizecurrentposewithoutIMU in tracker.cpp due to seg faults. I am still working on solving this.

grreyes2, 2019-07-09:
Curious if there have been any updates on this issue, I am encountering the same problem.

Doodle1106, 2019-09-05:
have you tried replacing optimizePoseFaster with optimizecurrentposewithoutIMU? It usually helps

MasterPa, 2019-10-17:
Issue closed, re-open if needed.

closed 2019-10-17 by MasterPa

Rindelwald, 2020-11-16:
I had the same problem. Anybody help？