generalized-intelligence/GAAS #31: Tutorial 4 not publishing Octomap

dan-riley, 2019-05-15:
I get through point cloud display in tutorial 4, but it's not publishing any octomap messages. Rviz sees the message, but nothing gets displayed. If I rostopic echo /occupied_cells_vis_array it says there's no messages and asked if /clock is being published. When I run Navigator, it crashes into the wall, presumably because it doesn't have an octomap? It has shows the point cloud, but I'm guessing it's using the octomap for planning.

Doodle1106, 2019-05-16:
1. The last few lines of your octomap_mapping.launch should be like:

                
                
                <remap from="cloud_in" to="cloud_in" />

Othewise Octomap will not be able to receive any information.

2. Yes, **occupied_cells_vis_array** gives occupied and unavailable cells that A* needs to find a path, without which the drone will quickly crashes into the wall as no obstacle information is given.

Please use the provided template next time if you spot any issue.

closed 2019-05-18 by Doodle1106

dan-riley, 2019-05-20:
Just wanted to add that I found the problem.  I had the above line in my octomap_mapping.launch, and I thought I had all of the other edits from the tutorial in as well, but somehow I missed the "frame_id" set to "/map".  Changing that fixed everything!  Thanks!

MasterPa, 2019-05-21:
Thanks for letting us know!