generalized-intelligence/GAAS #33: Problem about the compiling of SLAM in Tutorial 3.

pxfdale, 2019-05-22:
### I have a problem when I am running " $ sh generate.sh ", it seems that I had some problems in building the dependency of SLAM. ( But I have already followed the steps before...)

### The first position error exsists in my log is like this:

-- Checking for module 'openni-dev'
--   No package 'openni-dev' found
-- Found openni: /usr/lib/libOpenNI.so  
-- Checking for module 'openni2-dev'
--   No package 'openni2-dev' found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

### And then in this log errors are like this:

/home/pxf/catkin_ws/src/GAAS/software/SLAM/ygz_slam_ros/Thirdparty/DBow3/src/../src/Database.h:31:12: warning: ‘DBoW3::MIN_COMMON_WORDS’ defined but not used [-Wunused-variable]

### or like this:

/home/pxf/catkin_ws/src/GAAS/software/SLAM/ygz_slam_ros/backend/src/BackendSlidingWindowG2O.cpp:446:24: error: ‘make_unique’ is not a member of ‘g2o’

### What can I do to solve this problem? Or at least know what the real problem is?

### Thanks a lot for answering such a basic problem, but I have spend several days on it and still can not find the clues.

ninawrong, 2019-05-23:
Thanks for submitting the issue :) We will try to reciprocate the failure and get back to you.

Doodle1106, 2019-05-24:
how did you install g2o?

my guess is you were using an ancient version of G2O when g2o::make_unique did't exist.

Doodle1106, 2019-05-24:
try the master branch g2o and rebuild again, don't forget to make install.
The g2o has not been updated for quite a few months so a proper rebuild and installation should fix the problem.
Also warnings like that are not important to the performance.

closed 2019-05-25 by Doodle1106