generalized-intelligence/GAAS #35: combination of gps and slam navigation

guoxianzhen, 2019-05-29:
Please provide any relevant information about your setup.
* Tutorial No.:[Tutorial 3]
* Mavros: [0.18.4]
* ROS: [Kinetic]
* Gazebo: [7]
* Ubuntu: [16.04]
* System Platform: [X86]
 I have implemented the tutorial 3 for SLAM navigation. However, my final year projection is about combination of slam and gps navigation. I am wondering if i can use GAAS  for my fyp. 
I replace the modeled iris copter used in tutorial 3 with iris_stereo_gray_gps. Then SLAM is launched and EKF2_AID_MASK is configured to 9 which allows for gps and vision pose fusion. 
Then, the copter takes off under gps and slam navigation.  I close GPS with command "gpssim stop"  when the copter is hovering after taking off. The copter keep hovering without GPS and it can implement simple mission like fly 6 meter to the right then land under SLAM navigation.  So I think these steps may work.
But I have not tried any complicate simulation like closing GPS while copter is moving. I am not sure if the previous steps are right. Do I need to do any configuration after closing GPS?
## Screenshots
[image]
Here is the position information of my simulation. The dashed lines represent vision pose, The thin lines are local positions from estimator and the thick lines are local positions from GPS.

cyanine-gi, 2019-10-11:
Here we've got a branch called dev_gog.In this branch we have a whole set of logic to handle the fusion of unstable gps signal and visual-inertial slam info.You can check algorithms/GlobalOptimizationGraph_GTSAM to have a preview of this set of tools.Also there are some python scripts for you to check the path in visualized graphs.
We will release this module after well tested.

MasterPa, 2019-10-17:
Issue closed, re-open if needed.

closed 2019-10-17 by MasterPa