generalized-intelligence/GAAS #36: website redesigning

rishav30sh, 2019-05-30:
are you guys thinking of redesigning https://www.gaas.dev/ (it's already a great website, though ) ?
i got here through reddit, started reading documentation and your tutorials actually makes sense to me :) 
i was thinking of integrating https://gaas.gitbook.io/guide/ inside the official website. It would be much cleaner and would make a good starting point to beginner like me.
If you guys are interested, i would like to purpose a much cleaner new official website, along with documentation. (i like what you are doing, and just wanted to be part of this amazing family)

ninawrong, 2019-05-31:
Hi Ravish,

That's a great suggestion! I know the website is a bit simple as of now. We built it with Wix so the UX is prob not the best due to limited functionalities provided by Wix. 
We would love to see your design :) It will definitely provide us with some guidance when we rebuild the website.

Can you send me a scatch for the new design via nina.rong@gaas.dev? I don't want to take too much of your time so a simple illustration would do :)

Cheers,
Nina

ninawrong, 2019-06-05:
After talking to Rishav, we have decided to rebuild the website when the project is more mature.
I will close this issue for now. This issue can be re-opened when we come to the rebuilding of the website.

closed 2019-06-05 by ninawrong