generalized-intelligence/GAAS #38: How to use mavros Publish a XYZ data to put the aircraft in 3DFIX without GPS？

binbinshare, 2019-06-04:
geometry_msgs::PoseStamped vision;
int main(int argc, char **argv)
{
	ros::init(argc, argv, "uwb");
	ros::NodeHandle nh("~");
	ros::Publisher vision_pub = nh.advertise<geometry_msgs::PoseStamped>("/mavros/vision_pose/pose", 100);

	// 频率
	ros::Rate rate(20.0);
	while (ros::ok())
	{
		//回调一次 更新传感器状态
		ros::spinOnce();
		vision.pose.position.x = 116.3;
		vision.pose.position.y = 39.9;
		vision.pose.position.z = 3;

		vision.pose.orientation.x = 0.707107;
		vision.pose.orientation.y = 0;
		vision.pose.orientation.z = 0;
		vision.pose.orientation.w = 0.707107;

		vision.header.stamp = ros::Time::now();
		vision_pub.publish(vision);

		rate.sleep();
	}
	return 0;
}
I want to use this demo， put the aircraft in 3DFIX without GPS，but it not work。

Doodle1106, 2019-06-06:
1. First of all, it is not a GAAS problem but a MAVROS issue.
2. If I understand you correctly you wish to control the drone to a xyz target position without GPS with the help of vision information.  Check the third tutorial, which gives an example on how to set related parameters to enable vision fusion. And if you wish to use your own SLAM algorithm, you can refer to software/SLAM/ygz_slam_ros/examples/EurocStereoVIO_ros.cpp which illustrates how to convert raw slam output to vision_pose that px4 requires.
3. Regarding your example given above, firstly, you are not publishing control command to the right topic. This link gives a minimal working example: https://dev.px4.io/en/ros/mavros_offboard.html. Or you can refer to our first tutorial for some extra work. Secondly, vision information should be published to mavros/vision_pose/pose, and it should not be a constant value, otherwise the drone will be destined to clash.

In your case, refer to tutorial 1 for information on how to control your drone in GPS mode, then refer to tutorial 3 for how to enable SLAM in GPS denied environment.

Doodle1106, 2019-06-06:
有问题加微信群。

closed 2019-06-06 by Doodle1106

binbinshare, 2019-06-06:
thanks，i fixed it.