generalized-intelligence/GAAS #43: Obstacle_Map calibration mynteye2.yml question

lgh5054, 2019-06-24:
# Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [v ] I am running the latest version of GAAS
- [ v] I checked the Gitbook documentation and found no answer
- [ v] I checked to make sure that this issue has not already been filed
- [ ] I Googled/Binged the error messages generated by Ubuntu, Gazebo, and PX4
- [ ] I have posted questions regarding Ubuntu, Gazebo, and PX4 in corresponding forums.

# Issue Template

## Context
Please provide any relevant information about your setup.
* Tutorial No.:[Tutorial 4]
* Mavros: [ 0.18.4]
* ROS: [Kinetic]
* Gazebo: [7]
* Ubuntu: [16.04]
* System Platform: [jetson tx2]

Hello 

I'm using mynteyes-s camera.

This is the screen that me put into action

1.sh run.sh
2./bin/match_pointcloud_and_slam_pose2

[image]
you can see very small  dense point cloud appeared,

Is this the problem caused by the obstacle_map/calibration/mynteye2.yml file not matching my camera? 

Then, tell me how to convert my Yaml file to mynteye2.yml

Doodle1106, 2019-06-25:
1. in config.yaml, are you using rectified image or raw image? change left and right camera topics that ends with "image_rect"  if you are not using the right topics;
2.  that tiny point cloud could be the result of a poor parameter selection for:
    UseRadiusOutlierRemoval
which you can find in the bottom of config.yaml.
Try tuning Radius and MinNeighbor, they could give you a better result.

after the first two step, you should start getting meaningful point clouds, for better performance, follow step 3.

3. as far as I know, mynteye cameras provide a method to get instrinstic parameters including K, D, update correspondingly in camera config file. 

Let us know if you can get a meaningful point cloud.

lgh5054, 2019-07-01:
[image]

I'm testing GAAS software on real-drone.

 I connected the pixhawk and the companion computer.

 It activates the ygz-slam and checks the rviz to the vision position.

 But as you can see in the picture, it creates a map that is different from the image.

 What's the problem? I want to make a map exactly like the video GAAS uploaded on June 2. 

 Please help me

lgh5054, 2019-07-09:
The problem was the slam.

Close.

closed 2019-07-09 by lgh5054