generalized-intelligence/GAAS #44: [BUG]

lgh5054, 2019-07-09:
# Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ v ] I am running the latest version of GAAS
- [ v ] I checked the Gitbook documentation and found no answer
- [  v] I checked to make sure that this issue has not already been filed

# Issue Template

## Context
Please provide any relevant information about your setup.
* Mavros: [0.18.4]
* ROS: [Kinetic]
* Gazebo: [ 7]
* Ubuntu: [16.04]
* System Platform: [X86]

I installed the firmware version of px4fmu_v2 1.8.0 on GAAS hardware

And ekf2_aid_mask 8 
sys_companion (921600)
ekf2_eva_Noise 0.20
ekf2_evp_Noise 0.20
ekf2_ev_delay 175.0ms 
I watched the paramedics

When viewing the mavros/local_position/pose value, Even if you stay still, you lose your position.
Is this a slam problem or a connection speed problem with the pixhawk and tx2 board?

position value x y 0

If this is solved, everything will be solved. help me..

[image]
[image]

MasterPa, 2019-07-09:
This seems like a problem about connection, not about GAAS.

lgh5054, 2019-07-09:
Can you tell how Gaas set up communication?

closed 2019-07-09 by lgh5054