generalized-intelligence/GAAS #49: Ubuntu 18.04 import for ros melodic? As jetson nano only supports 18.04!

atmadeep, 2019-08-19:
Hi, I'm making an autonomous drone using Jetson nano. The Jetson nano Sdk is based on Ubuntu 18.04. So how much of this code will work with ROS Melodic Morenia? Are you guys working on porting this to Ubuntu 18.04?

ninawrong, 2019-08-20:
Several developers have informed us that they failed to run GAAS on Jetson nano as GAAS requires a computer with higher performance.

Though we do not support ROS Melodic Merenia with Ubuntu 18.04, one of the contributors is developing under this environment. We know it's doable but you may need to play with the configuration for a bit.

We are currently not working on porting this to Ubuntu 18.04.

atmadeep, 2019-08-20:
Hi, So your build has a TX1 or I guess a RPi as a companion computer. But I
think we can save a lot of money on TX1 and add way more functionality than
a RPi if we use jetosn nano. I'm working on such a project myself. This
way, we can have a cheaper test ground for testing out vision and fusion
algorithms. As you rightly mentioned, GAAS cannot run on a Jetson Nano, but
Some vision algorithms can. I'll update you more on this.
But can you guys help me find a way to connect a pixhawk to a Jetson Nano?
I think it'll probably be same like RPi 3.
Can you help me with that?

hddgi, 2019-08-21:
Hey, @atmadeep , in theory, you only need to follow the [GAAS tutorial1](https://gaas.gitbook.io/guide/software-realization-build-your-own-autonomous-drone/build-your-own-autonomous-drone-e01-offboard-control-and-gazebo-simulation). After installing MAVLink and MAVROS, you can connect pixhawk, and use the script control provided by [GAAS demo](https://github.com/generalized-intelligence/GAAS/tree/master/demo/tutorial_1/1_px4_mavros_offboard_controller).

closed 2019-08-29 by ninawrong

Lndulge666, 2020-10-22:
hey, @atmadeep ,did you successfully use GAAS on ubuntu 18.04?