generalized-intelligence/GAAS #52: ModuleNotFoundError: No module named 'bresenham3d'

machenxiang, 2019-09-03:
* Tutorial No.:Tutorial 4
* Mavros: 0.18.4
* ROS: Kinetic
* Gazebo: 7
* Ubuntu: 16.04
* System Platform: X86
* python :Python 3.7.3 (default, Mar 27 2019, 22:11:17) 
[GCC 7.3.0] :: Anaconda, Inc. on linux

When python Navigator.py  it occurs this problem.
Traceback (most recent call last):
  File "Navigator.py", line 24, in <module>
    from path_optimization.path_pruning import PathPruning
  File "/home/mcx/GAAS/software/Navigator/path_optimization/path_pruning.py", line 3, in <module>
    from bresenham3d import Bresenham3D
ModuleNotFoundError: No module named 'bresenham3d'

I have google this problem. But i just find bresenham no bresenham3d. How can i solve this problem? Thanks in advance.

metaphox, 2019-09-03:
The package `bresenham3d` is just the file `bresenham3d.py` which should be found under `/home/mcx/GAAS/software/Navigator/path_optimization`, the same directory as `path_pruning.py`. Please make sure you have this file in your system.

machenxiang, 2019-09-10:
> The package `bresenham3d` is just the file `bresenham3d.py` which should be found under `/home/mcx/GAAS/software/Navigator/path_optimization`, the same directory as `path_pruning.py`. Please make sure you have this file in your system.

Thank you

closed 2019-10-11 by cyanine-gi