generalized-intelligence/GAAS #53: [BUG] E06: Basic Object Tracking,When running  track_and_move.py

zjuxlx, 2019-09-04:
## Context
My relevant information about my setup.

* ROS: [Kinetic]
* Gazebo: [7.15.0]
* Ubuntu: [16.04]
* System Platform: [X86]
* Python: [2.7.12]

## Current Behavior
Autonomous Drone Software E06: Basic Object Tracking

1.

When python track_and_move.py It occurs this question:
Traceback (most recent call last):
  File "track_and_move.py", line 134, in <module>
    SimpleTrackAndMove(resolution, K,left_topic=left_topic, right_topic=right_topic, object_position_topic=object_position_topic)
TypeError: __init__() takes at least 4 arguments (6 given)
[image]
The source code is:
[image]
[image]

2.

When I delete commander in def __init__(),
[image]
It occurs this question:
[image]

3.

And if I  add commander in the main function,
[image]

It occurs this question:
[image]

metaphox, 2019-09-04:
Please try the following:

1. remove the whole line 126 from `track_and_move.py`, i.e. remove `rospy.init_node('track_and_move', anonymous=True)`, since the `Commander` class will call `init_node` on its own.
2. remove the `commander` argument from the `SimpleTrackAndMove` call on line 134.

zjuxlx, 2019-09-05:
Thanks for your reply.
I made a success after making this following changes in track_and_move.py
1. remove the commander argument from the def __inti__() on line 20.
2. remove the whole line 126.
3. remove the commander argument frome the SimpleTrackAndMove call on line 134.

ninawrong, 2019-09-05:
Issue closed as the problem is solved

closed 2019-09-05 by ninawrong