generalized-intelligence/GAAS #54: Failed to load plugin libgazebo

robin-shaun, 2019-09-09:
When I `roslaunch px4 path_planning.launch`, errors occur.
```
[Err] [Plugin.hh:165] Failed to load plugin libgazebo_multirotor_base_plugin.so: libgazebo_multirotor_base_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin libgazebo_gps_plugin.so: libgazebo_gps_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin libgazebo_mavlink_interface.so: libgazebo_mavlink_interface.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin libgazebo_imu_plugin.so: libgazebo_imu_plugin.so: cannot open shared object file: No such file or directory
```
However all about gazebo have been installed.
```
gazebo7/xenial,now 7.0.0+dfsg-2 amd64 [已安装，自动]
  Open Source Robotics Simulator - Binaries

gazebo7-common/xenial,xenial,now 7.0.0+dfsg-2 all [已安装，自动]
  Open Source Robotics Simulator - Shared files

gazebo7-doc/xenial,xenial 7.0.0+dfsg-2 all
  Open Source Robotics Simulator - Documentation

gazebo7-plugin-base/xenial,now 7.0.0+dfsg-2 amd64 [已安装，自动]
  Open Source Robotics Simulator - base plug-ins

libgazebo7/xenial,now 7.0.0+dfsg-2 amd64 [已安装，自动]
  Open Source Robotics Simulator - shared library

libgazebo7-dev/xenial,now 7.0.0+dfsg-2 amd64 [已安装，自动]
  Open Source Robotics Simulator - Development Files

ros-kinetic-gazebo-dev/xenial,now 2.5.19-1xenial-20190607-165824-0800 amd64 [已安装]
  Provides a cmake config for the default version of Gazebo for the ROS distribution.

ros-kinetic-gazebo-msgs/xenial,now 2.5.19-1xenial-20190608-051945-0800 amd64 [已安装]
  Message and service data structures for interacting with Gazebo from ROS.

ros-kinetic-gazebo-plugins/xenial,now 2.5.19-1xenial-20190608-115731-0800 amd64 [已安装]
  Robot-independent Gazebo plugins for sensors, motors and dynamic reconfigurable components.

ros-kinetic-gazebo-ros/xenial,now 2.5.19-1xenial-20190608-134931-0800 amd64 [已安装]
  Provides ROS plugins that offer message and service publishers for interfacing with Gazebo through ROS.

ros-kinetic-gazebo-ros-control/xenial,now 2.5.19-1xenial-20190608-140610-0800 amd64 [已安装]
  gazebo_ros_control

ros-kinetic-gazebo-ros-pkgs/xenial,now 2.5.19-1xenial-20190608-140634-0800 amd64 [已安装]
  Interface for using ROS with the Gazebo simulator.
```

robin-shaun, 2019-09-09:
The problem has been solved, some directories in .bashrc are wrong.

closed 2019-09-09 by robin-shaun

ashu41, 2019-09-24:
How did you solve it? I'm facing the same issue

robin-shaun, 2019-09-25:
> How did you solve it? I'm facing the same issue

Please make sure all the paths about ROS are correct in .bashrc.