generalized-intelligence/GAAS #55: Tutorisas4 question，python Navigator occur errors.

machenxiang, 2019-09-10:
* Tutorial No.:[ Tutorial 4]
* Mavros: [0.18.4]
* ROS: [Kinetic]
* Gazebo: [7]
* Ubuntu: [16.04]
* System Platform: [ X86]

## Current Behavior

When I python Navigator.py. It occurs an error. When i use source activate py2.7 to change python version to python2.7,It occurs this problem.

Traceback (most recent call last):
  File "Navigator.py", line 422, in <module>
    nav = Navigator()
  File "Navigator.py", line 89, in __init__
    self.set_status(status.INITIALIZED)
  File "Navigator.py", line 401, in set_status
    self.STATUS = String(status.name)
AttributeError: 'int' object has no attribute 'name'

But when i use Python 3.7，it ocuurs the different error.
Traceback (most recent call last):
  File "Navigator.py", line 22, in <module>
    from path_optimization.path_pruning import PathPruning
  File "/home/mcx/GAAS/software/Navigator/path_optimization/path_pruning.py", line 4, in <module>
    from bresenham3d import Bresenham3D
ModuleNotFoundError: No module named 'bresenham3d'
[image]
[image]

machenxiang, 2019-09-14:
I have solved this problem. I think the reason for this question is the version of the enum is too old. There are two ways to solve this question.
1. You just need to use new python.
2 If you are using python2.7 you can pip install enum34.

MasterPa, 2019-09-14:
Issue closed as the problem is solved.

closed 2019-09-14 by MasterPa