generalized-intelligence/GAAS #56: No point cloud in rviz

ChenChangHaoWS, 2019-09-16:
# Issue Template

## Context
Please provide any relevant information about your setup.
* Tutorial No.:Tutorial 4
* Mavros: 0.32.1
* ROS: Kinetic
* Gazebo: 7
* Ubuntu: 16.04
* System Platform: X86

## Expected Behavior

When doing the
[image]
I should see point cloud in rviz like
[image]

## Current Behavior

I can't see any point cloud in rviz.

## Screenshots
I just do as the tutorials told, everything I can see in my screen:
[image]

## Failure Logs

Copy console log and paste it here.
[image]
[image]

And the 'path_planning.launch' consloe log:
... logging to /home/ubuntu/.ros/log/dbb5e6f4-d824-11e9-8baa-03af469f68e4/roslaunch-cch-38852.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cch:33805/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
[... 343 lines cut ...]
[ INFO] [1568598849.755910592, 999.412000000]: VER: 1.1: Flight software:     010800ff (96443b3cf3e3adae)
[ INFO] [1568598849.755988852, 999.412000000]: VER: 1.1: Middleware software: 010800ff (96443b3cf3e3adae)
[ INFO] [1568598849.756041332, 999.412000000]: VER: 1.1: OS software:         040f00ff (0000000000000000)
[ INFO] [1568598849.756087072, 999.412000000]: VER: 1.1: Board hardware:      00000001
[ INFO] [1568598849.756137488, 999.412000000]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1568598849.756228992, 999.412000000]: VER: 1.1: UID:                 0000000100000002
[ WARN] [1568598849.761542620, 999.418000000]: CMD: Unexpected command 520, result 0
INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF commencing GPS fusion
[ INFO] [1568598858.773406476, 1008.408000000]: HP: requesting home position
[ INFO] [1568598863.783141454, 1013.408000000]: WP: mission received
[mavros-6] killing on exit
[gazebo_gui-4] killing on exit
[gazebo-3] killing on exit
[sitl-2] killing on exit

Exiting...
pxh> Shutting down
Restoring terminal
[gazebo-3] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

cyanine-gi, 2019-10-11:
You may use rostopic hz /cloud_in to check if there are any msgs publishing.
And you may check your rviz configuration, especially the Global Options-> Fixed Frame.

MasterPa, 2019-10-17:
Issue closed, re-open if needed.

closed 2019-10-17 by MasterPa