generalized-intelligence/GAAS #58: Vehicle Took Off Failed!

Manakoy, 2019-09-28:
`fyh@A:~/GAAS/demo/tutorial_1/1_px4_mavros_offboard_controller$ python px4_mavros_run.py
Px4 Controller Initialized!
Vehicle Took Off Failed!`

`[ WARN] [1569651901.031346966, 13.780000000]: PR: request param #79 timeout, retries left 2, and 1 params still missing
[ WARN] [1569651901.033471004, 13.780000000]: PR: Param COM_EF_TIME (78/563): <value><double>10</double></value> different index: 79/564
[ INFO] [1569651904.560061990, 17.300000000]: WP: mission received
[ INFO] [1569651931.280981252, 43.960000000]: FCU: ARMED by Arm/Disarm component command
[ WARN] [1569651931.288244403, 43.968000000]: CMD: Unexpected command 176, result 1
INFO  [commander] Takeoff detected
[ INFO] [1569651932.480372666, 45.156000000]: FCU: Takeoff detected
WARN  [commander] Failsafe enabled: no RC and no offboard
[ERROR] [1569651933.720366873, 46.396000000]: FCU: Failsafe enabled: no RC and no offboard`

What caused this problem? I took off successfully the first time, but then I got this error.

Doodle1106, 2019-10-09:
Sorry for such a late response, but have you tried restarting simulation?

MasterPa, 2019-10-17:
Issue closed, re-open if needed.

closed 2019-10-17 by MasterPa

cillian-bao, 2021-01-04:
I also have this problem