generalized-intelligence/GAAS #63: issue when i try to use slam to build 3d model

onionysy, 2019-10-18:
# Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [yes ] I am running the latest version of GAAS
- [yes ] I checked the Gitbook documentation and found no answer
- [yes ] I checked to make sure that this issue has not already been filed
- [yes ] I Googled/Binged the error messages generated by Ubuntu, Gazebo, and PX4
- [yes ] I have posted questions regarding Ubuntu, Gazebo, and PX4 in corresponding forums.

# Issue Template

## Context
Please provide any relevant information about your setup.
* Tutorial No.:Tutorial 2
* Mavros: forgot
* ROS: Kinetic
* Gazebo: 7
* Ubuntu: 16.04
* System Platform: X86

## Expected Behavior

In your tutorial, after running rostopic list, there are 2 topics called /gi/simulation/XX/image_raw.
But, I follow your tutorial on gitbook and build from source, when I get into the part2, I can't find anything about the camera and I don't know how to use the slam and SfM simulation.
Then I use the docker way for making sure that I didn't miss any tiny. But, still, when I tap in 'rostopic list', /gi doesn't show up.

Did I forget building the "gi package" or something related?

I see it has been raised months age, and you said you will fixed it. But I still been stuck in this. Can you give me some advices? THX

跟着gitbook的教程走，part1 还好说，part2的时候卡住了，毫无头绪，斗胆来issue提问下。找不到相机数据那个topic，是不是得自己设置相机。还有如何安装，调用slam模块?
这个问题之前看别人也提过，据当时说过一阵会被修复。但我还是在这里出现了同样的问题。能麻烦您帮忙解答下我该如何操作么？
## Current Behavior
ROS TOPIC
`/clock
/diagnostics
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/gazebo_gui/parameter_descriptions
/gazebo_gui/parameter_updates
/mavlink/from
/mavlink/to
/mavros/actuator_control
/mavros/adsb/send
/mavros/adsb/vehicle
/mavros/altitude
/mavros/battery
/mavros/cam_imu_sync/cam_imu_stamp
/mavros/companion_process/status
/mavros/debug_value/debug
/mavros/debug_value/debug_vector
/mavros/debug_value/named_value_float
/mavros/debug_value/named_value_int
/mavros/debug_value/send
/mavros/extended_state
/mavros/fake_gps/mocap/tf
/mavros/global_position/compass_hdg
/mavros/global_position/global
/mavros/global_position/gp_lp_offset
/mavros/global_position/gp_origin
/mavros/global_position/home
/mavros/global_position/local
/mavros/global_position/raw/fix
/mavros/global_position/raw/gps_vel
/mavros/global_position/raw/satellites
/mavros/global_position/rel_alt
/mavros/global_position/set_gp_origin
/mavros/gps_rtk/send_rtcm
/mavros/hil/actuator_controls
/mavros/hil/controls
/mavros/hil/gps
/mavros/hil/imu_ned
/mavros/hil/optical_flow
/mavros/hil/rc_inputs
/mavros/hil/state
/mavros/home_position/home
/mavros/home_position/set
/mavros/imu/data
/mavros/imu/data_raw
/mavros/imu/diff_pressure
/mavros/imu/mag
/mavros/imu/static_pressure
/mavros/imu/temperature_baro
/mavros/imu/temperature_imu
/mavros/landing_target/lt_marker
/mavros/landing_target/pose
/mavros/landing_target/pose_in
/mavros/local_position/accel
/mavros/local_position/odom
/mavros/local_position/pose
/mavros/local_position/pose_cov
/mavros/local_position/velocity_body
/mavros/local_position/velocity_body_cov
/mavros/local_position/velocity_local
/mavros/log_transfer/raw/log_data
/mavros/log_transfer/raw/log_entry
/mavros/manual_control/control
/mavros/manual_control/send
/mavros/mission/reached
/mavros/mission/waypoints
/mavros/mocap/pose
/mavros/mount_control/command
/mavros/obstacle/send
/mavros/odometry/in
/mavros/odometry/out
/mavros/param/param_value
/mavros/px4flow/ground_distance
/mavros/px4flow/raw/optical_flow_rad
/mavros/px4flow/raw/send
/mavros/px4flow/temperature
/mavros/radio_status
/mavros/rc/in
/mavros/rc/out
/mavros/rc/override
/mavros/setpoint_accel/accel
/mavros/setpoint_attitude/cmd_vel
/mavros/setpoint_attitude/thrust
/mavros/setpoint_position/global
/mavros/setpoint_position/local
/mavros/setpoint_raw/attitude
/mavros/setpoint_raw/global
/mavros/setpoint_raw/local
/mavros/setpoint_raw/target_attitude
/mavros/setpoint_raw/target_global
/mavros/setpoint_raw/target_local
/mavros/setpoint_velocity/cmd_vel
/mavros/setpoint_velocity/cmd_vel_unstamped
/mavros/state
/mavros/statustext/recv
/mavros/statustext/send
/mavros/target_actuator_control
/mavros/time_reference
/mavros/timesync_status
/mavros/trajectory/desired
/mavros/trajectory/generated
/mavros/trajectory/path
/mavros/vfr_hud
/mavros/vision_pose/pose
/mavros/vision_pose/pose_cov
/mavros/vision_speed/speed_twist_cov
/mavros/wind_estimation
/rosout
/rosout_agg
/tf
/tf_static
`

## Screenshots
If applicable, add screenshots to help explain your problem.

Copy console log and paste it here.

onionysy, 2019-10-18:
And when I try to use the slam, it can not visualize left camera image in real time. It is only the dark screen./home/comac/图片/2019-10-18 16-29-44屏幕截图.png

onionysy, 2019-10-18:
when i roslaunch, it raise this
`comac@comac-HP-Z840-Workstation:~/GAAS/software/Obstacle_Map$ roslaunch px4 path_planning.launch
... logging to /home/comac/.ros/log/c85b50a4-f182-11e9-ac30-44334c49fbe0/roslaunch-comac-HP-Z840-Workstation-19074.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://comac-HP-Z840-Workstation:42703/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
 * /mavros/global_position/tf/send: False
[... 308 lines cut ...]
[ INFO] [1571388131.052932373, 998.392000000]: IMU: High resolution IMU detected!
[ INFO] [1571388131.195906029, 998.536000000]: IMU: Attitude quaternion IMU detected!
[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/iris_stereo_gray_gps/motor_speed/3, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/iris_stereo_gray_gps/imu, deleting message. This warning is printed only once.
[ INFO] [1571388132.056172691, 999.394000000]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1571388132.056279553, 999.394000000]: VER: 1.1: Flight software:     010800ff (96443b3cf3e3adae)
[ INFO] [1571388132.056336686, 999.394000000]: VER: 1.1: Middleware software: 010800ff (96443b3cf3e3adae)
[ INFO] [1571388132.056399223, 999.394000000]: VER: 1.1: OS software:         040f00ff (0000000000000000)
[ INFO] [1571388132.056452638, 999.394000000]: VER: 1.1: Board hardware:      00000001
[ INFO] [1571388132.056526471, 999.394000000]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1571388132.056571226, 999.394000000]: VER: 1.1: UID:                 0000000100000002
[ WARN] [1571388132.104512469, 999.440000000]: CMD: Unexpected command 520, result 0
INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF commencing GPS fusion
[ INFO] [1571388141.069492081, 1008.390000000]: HP: requesting home position
[ INFO] [1571388146.076236950, 1013.390000000]: WP: mission received
ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[ERROR] [1571388255.475664445, 1122.634000000]: TM : Time jump detected. Resetting time synchroniser.
ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
[ERROR] [1571388488.856532432, 1355.734000000]: TM : Time jump detected. Resetting time synchroniser.
`

I think the reason maybe this
 `[Err] [Plugin.hh:165] Failed to load plugin libgazebo_ros_multicamera.so: libgazebo_ros_multicamera.so: cannot open shared object file: No such file or directory`
But I don't know how to fix it. Can you help me?

Doodle1106, 2019-10-21:
Sorry for the late response.

It seems like a misconfiguration of ros-gazebo-pkg, can you try the following command?

`sudo apt install ros-kinetic-gazebo-ros-pkgs`

onionysy, 2019-10-21:
thx for your guide. I have fix it. thank you.

closed 2019-10-21 by onionysy