generalized-intelligence/GAAS #65: issues with displaying video of drone

Arsalan66, 2019-10-20:
# Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version of GAAS
- [no answer ] I checked the Gitbook documentation and found no answer
- [ i checked all resources] I checked to make sure that this issue has not already been filed
- [ i have searched a lot] I Googled/Binged the error messages generated by Ubuntu, Gazebo, and PX4
- [ only one forum] I have posted questions regarding Ubuntu, Gazebo, and PX4 in corresponding forums.

# Issue Template

## Context
Please provide any relevant information about your setup.
* Tutorial No.:[all]
* Mavros: [eg. 0.18.4]
* ROS: [ melodic]
* Gazebo: [ 7]
* Ubuntu: [eg. 18.04]
* System Platform: [ X86]

## Expected Behavior

Please describe the behavior you are expecting.
I want to process videos based on drone's camera
## Current Behavior

What is the current behavior?
I have created a publisher and subscriber to video frames on a custom video and it runs perfectly however when i subscribe to topics related to camera, it doesn't show up anything although my program says that frames are being received
## Screenshots
If applicable, add screenshots to help explain your problem.
here's a screenshot which runs my code
[image]
[image]
here's the output of custom video
[image]

## Failure Logs
this is the topic that i subscribe to for drone, it says frames are being recieved but doesn't show anything
[image]
i made a custom drone:
[image]

Copy console log and paste it here.

Arsalan66, 2019-10-20:
what are the steps that i'm doing wrong and what can i do to rectify them?

Doodle1106, 2019-10-21:
I do not think the problem is related to this repo.

If you are sure callbacks are executed, you can add some print functions to help you debug,  I don't quite see any issue in the code you provided.

Arsalan66, 2019-10-21:
Thanks for the kind response basically I want to implement IBVS via colour detection schemes suppose if I do this in GAAS framework, what are the subscribers that I would have to keep in mind to manipulate the camera, I'll be grateful for your kind response

Arsalan66, 2019-10-21:
i ALSO usec cv2.imshow to check

hddgi, 2019-10-24:
Can you confirm that you have published information by using the “rostopic” command? Or can you display your published Image msg in ”rviz“?

Arsalan66, 2019-10-24:
[image]
i made a custom publisher but in case of manipulating the drone's camera i'm unsure about the subscribers

Arsalan66, 2019-10-24:
sir can i have your mail, i want to discuss this small issue with you

Arsalan66, 2019-10-24:
Ok sir , I'll provide as soon as possible , thanks for the kind response

Arsalan66, 2019-10-31:
#!/usr/bin/env python

import rospy, cv2
#from std_msgs.msg import String
import numpy as np
from sensor_msgs.msg import Image
from cv_bridge import CvBridge, CvBridgeError

bridge = CvBridge()
def tracking_ball(mask,cv_image):
    # create a black image
    black_image = np.zeros(cv_image.shape, np.uint8)
    _,contours,_ = cv2.findContours(mask,cv2.RETR_LIST,cv2.CHAIN_APPROX_SIMPLE)
    for c in contours:
        if cv2.contourArea(c) > 1000:
            # find radius
            ((x,y),r) = cv2.minEnclosingCircle(c)
            # find the cx, cy of the contour
            m = cv2.moments(c)
            cx = -1
            cy = -1
            if m['m00'] != 0:
                cx = int(m['m10']/m['m00'])
                cy = int(m['m01']/m['m00'])
            # draw the contours
            cv2.drawContours(cv_image,[c],-1,(0,0,255),1)
            cv2.drawContours(black_image,[c],-1,(0,0,255),1)
            # draw a circle around the ball
            cv2.circle(cv_image,(cx,cy),int(r),(255,0,0),1)
            cv2.circle(black_image,(cx,cy), int(r),(255,0,0),1)
            # draw the center of the ball
            cv2.circle(cv_image,(cx,cy),5,(0,0,0),-1)
            cv2.circle(black_image,(cx,cy),5,(0,0,0),-1)
            # show images
            #img_join = np.concatenate((cv_image,black_image),axis=1)
            cv2.imshow('cv_image', cv_image)
            cv2.imshow('black_image',black_image)
            

def create_mask(hsv_image):
    lower_hsv = (30,100,100)
    upper_hsv = (50,255,255)
    mask = cv2.inRange(hsv_image,lower_hsv,upper_hsv)
    return mask

def hi_callback(ros_image):
  print 'got an image'
  global bridge
  cv_image = bridge.imgmsg_to_cv2(ros_image, "bgr8")
  hsv = cv2.cvtColor(cv_image,cv2.COLOR_BGR2HSV)
  mask=create_mask(hsv)
  tracking_ball(mask,cv_image)
  rospy.loginfo('frame received!')
  cv2.waitKey(5)

if __name__ == '__main__':
    
    # create a node
    rospy.init_node('tennis_ball_listener_node',anonymous=True)
    # create the subscriber
    rospy.Subscriber('/gi/simulation/left/image_raw',Image,hi_callback)
    # start a loop
    rospy.spin()

@hddgi  sir, this is my code , i want to process video published by '/gi/simulation/left/image_raw' topic, however the output was the same that frames are being received but not shown, what might i be doing wrong?

Arsalan66, 2019-10-31:
Sir, another method that i have in mind is to manipulate the publisher, where might be it's file located and is it a good idea or not?

Arsalan66, 2019-10-31:
here's the publisher that i used for testing
#!/usr/bin/env python 
import numpy as np
import cv2
from std_msgs.msg import String
from sensor_msgs.msg import Image
import rospy
import sys
from cv_bridge import CvBridge, CvBridgeError

bridge = CvBridge()

def talker():
    pub = rospy.Publisher('tennis_ball_image', Image, queue_size=10)
    rospy.init_node('talker', anonymous=True)
     # 1hz
    video_capture = cv2.VideoCapture('video/tennis-ball-video.mp4')
    rate = rospy.Rate(1)
    
    while not rospy.is_shutdown():
        # read a frame
        ret, frame = video_capture.read()
        #frame = cv2.resize(frame,None,fx=0.25,fy=0.25,interpolation=cv2.INTER_CUBIC)
        if ret:
            # publish frame in data
            frame_msg = bridge.cv2_to_imgmsg(frame,'bgr8')
            pub.publish(frame_msg)
            rospy.loginfo('Sending frame')
            rate.sleep()
    
    video_capture.release()
    cv2.destroyAllWindows()

if __name__ == '__main__':
    try:
        talker()
    except rospy.ROSInterruptException:
        pass

   
[image]

closed 2019-11-10 by Arsalan66