generalized-intelligence/GAAS #66: the Navigator can not work as image

onionysy, 2019-10-21:
# Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ yes] I am running the latest version of GAAS
- [ yes] I checked the Gitbook documentation and found no answer
- [ yes] I checked to make sure that this issue has not already been filed
- [ yes] I Googled/Binged the error messages generated by Ubuntu, Gazebo, and PX4
- [ yes] I have posted questions regarding Ubuntu, Gazebo, and PX4 in corresponding forums.

# Issue Template

## Context
Please provide any relevant information about your setup.
* Tutorial No.:[eg. Tutorial 4]
* Mavros: [eg. 0.18.4]
* ROS: [eg. Kinetic]
* Gazebo: [eg. 7]
* Ubuntu: [eg. 16.04]
* System Platform: [eg. X86]

I want to use Tutorial 4. 
I can  launch the simulation .
`comac@comac-HP-Z840-Workstation:~$ roslaunch px4 path_planning.launch
... logging to /home/comac/.ros/log/391b6068-f3cc-11e9-94fa-44334c49fbe0/roslaunch-comac-HP-Z840-Workstation-16585.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://comac-HP-Z840-Workstation:37541/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
[... 353 lines cut ...]
started roslaunch server http://comac-HP-Z840-Workstation:44033/

SUMMARY
========

PARAMETERS
 * /octomap_server/frame_id: /map
 * /octomap_server/resolution: 0.1
 * /octomap_server/sensor_model/max_range: 10
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    octomap_server (octomap_server/octomap_server_node)

ROS_MASTER_URI=http://localhost:11311

process[octomap_server-1]: started with pid [18694]

`
[image]

Hower I can not see the navi_path_plan topic in RVIZ.
And when I run the python Navigator.py.There are something wrong and the plane can not avoid the wall but crash on the wall. I think there are something wrong with it but I can find it, I hope you can help me with that. Thank you!

onionysy, 2019-10-21:
the navi_path_plan topic showed after i use `python Navigator.py` but it can not find a path and the plane crash on the wall.

onionysy, 2019-10-21:
I think it must be something wrong with that. If you think there are something unclear, can I have you wechat? THX

MasterPa, 2019-10-21:
You could join our wechat group by following our intro at the readme. However, we prefer to answer questions at GitHub issue.

onionysy, 2019-10-21:
> You could join our wechat group by following our intro at the readme. However, we prefer to answer questions at GitHub issue.
thx，I want to solve the problem at GITHUB ,too. Can you give me some advice about my question in running Navigation.py? Thank you very much.

commit f3b62b27 (2019-11-11): Fix tutorial 5 Navigator crash wall bug and coords bug, fixes #66.
closed 2019-11-11 by hddgi