generalized-intelligence/GAAS #67: How to use code in image detection(tegu_detection_demo.py)

Arsalan66, 2019-10-21:
# Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version of GAAS
- [ ] I checked the Gitbook documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I Googled/Binged the error messages generated by Ubuntu, Gazebo, and PX4
- [ ] I have posted questions regarding Ubuntu, Gazebo, and PX4 in corresponding forums.

# Issue Template

## Context
Please provide any relevant information about your setup.
* Tutorial No.:[all]
* Mavros: [ 0.18.4]
* ROS: [melodic]
* Gazebo: [. 7]
* Ubuntu: [eg. 16.04]
* System Platform: [eg. X86]

## Expected Behavior
I want to apply image processing techniques on drone's camera in GAAS, how can i use the code tegu_detection_demo.py to achieve my techniques, what are the subscribers to the drone's camera? And can i process the images as color based images using the drone's, my aim is to apply ibvs based on color

 schemes, help would really be appreciated 
regards 
Expected behavior is somewhat like this in the picture. this is on a ball i want to do this in gazebo on drone's camera that can be viewed 
[image]

qratosone, 2019-10-29:
Just look at tutorial 2

https://gaas.gitbook.io/guide/software-realization-build-your-own-autonomous-drone/build-your-own-autonomous-drone-part-2-build-a-3d-model-of-a-building-with-your-drone

The topic is ：

> /gi/simulation/left/image_raw

However, I don't think the image quality from gazebo is enough to train a deep learning model, you may need to use Airsim instead

Arsalan66, 2019-10-29:
Sir, thank you so much for the kind response my mid exams for 5th semester shall end this friday, i'll surely check this out and let you know the output

Arsalan66, 2019-10-31:
#!/usr/bin/env python

import rospy, cv2
#from std_msgs.msg import String
import numpy as np
from sensor_msgs.msg import Image
from cv_bridge import CvBridge, CvBridgeError

bridge = CvBridge()
def tracking_ball(mask,cv_image):
    # create a black image
    black_image = np.zeros(cv_image.shape, np.uint8)
    _,contours,_ = cv2.findContours(mask,cv2.RETR_LIST,cv2.CHAIN_APPROX_SIMPLE)
    for c in contours:
        if cv2.contourArea(c) > 1000:
            # find radius
            ((x,y),r) = cv2.minEnclosingCircle(c)
            # find the cx, cy of the contour
            m = cv2.moments(c)
            cx = -1
            cy = -1
            if m['m00'] != 0:
                cx = int(m['m10']/m['m00'])
                cy = int(m['m01']/m['m00'])
            # draw the contours
            cv2.drawContours(cv_image,[c],-1,(0,0,255),1)
            cv2.drawContours(black_image,[c],-1,(0,0,255),1)
            # draw a circle around the ball
            cv2.circle(cv_image,(cx,cy),int(r),(255,0,0),1)
            cv2.circle(black_image,(cx,cy), int(r),(255,0,0),1)
            # draw the center of the ball
            cv2.circle(cv_image,(cx,cy),5,(0,0,0),-1)
            cv2.circle(black_image,(cx,cy),5,(0,0,0),-1)
            # show images
            #img_join = np.concatenate((cv_image,black_image),axis=1)
            cv2.imshow('cv_image', cv_image)
            cv2.imshow('black_image',black_image)
            

def create_mask(hsv_image):
    lower_hsv = (30,100,100)
    upper_hsv = (50,255,255)
    mask = cv2.inRange(hsv_image,lower_hsv,upper_hsv)
    return mask

def hi_callback(ros_image):
  print 'got an image'
  global bridge
  cv_image = bridge.imgmsg_to_cv2(ros_image, "bgr8")
  hsv = cv2.cvtColor(cv_image,cv2.COLOR_BGR2HSV)
  mask=create_mask(hsv)
  tracking_ball(mask,cv_image)
  rospy.loginfo('frame received!')
  cv2.waitKey(5)

if __name__ == '__main__':
    
    # create a node
    rospy.init_node('tennis_ball_listener_node',anonymous=True)
    # create the subscriber
    rospy.Subscriber('/gi/simulation/left/image_raw',Image,hi_callback)
    # start a loop
    rospy.spin()

sir,this is my code, i tried to read video of tutorial 2 but it kept saying that frames are recieved but doesn't show, can you kindly tell me where publisher might be located?

closed 2019-11-10 by Arsalan66