generalized-intelligence/GAAS #74: [BUG] "move" behavior is not working rightly.

dsyou20, 2019-12-01:
# Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [* ] I am running the latest version of GAAS
- [* ] I checked the Gitbook documentation and found no answer
- [ *] I checked to make sure that this issue has not already been filed

# Issue Template

## Context
Please provide any relevant information about your setup.
* Mavros: [0.18.4]
* ROS: [Kinetic]
* Gazebo: [7]
* Ubuntu: [16.04]
* System Platform: [X86]

## Expected Behavior

when "move" command is issued,   the drone have to go forward directly.

## Current Behavior

whenever "move" command is issued,  the drone on px4 gazebo simulator follows the actions.
1. first slightly left turn
2. and then go forward 

# I used "px4_mavros_run.py" and "commander.py".
# here are code snapshot 
from commander import Commander
con = Commander()
con.move( 10, 0, 0 )
con.move( 10, 0, 0 )
con.move( 10, 0, 0 )

## Screenshots
If applicable, add screenshots to help explain your problem.

## Failure Logs

Copy console log and paste it here.

qratosone, 2019-12-01:
first we need to confirm if you are using GPS or SLAM

dsyou20, 2019-12-01:
No. It was tested on px4 gazebo simulator.

Note that if current_heading  from imu callback is disabled and the
direction of the head is fixed. it is working well.

Nevertheless, still orientation of head is not correct. for example,
 turn( 0 ) is not pointing out east.

Thanks

dsyou20, 2019-12-01:
tutorial 3 is used !

hddgi, 2019-12-03:
Which version of the px4 firmware do you use? We recommend you use px4 1.8.0 currently.

MasterPa, 2020-02-24:
Close the issue, reopen if needed.

closed 2020-02-24 by MasterPa