generalized-intelligence/GAAS #85: 请教：在E04实验中，rviz软件 有/gi 目录，1.这个目录是GAAS里的吗？2.请问如果是，rviz软件怎样知道GAAS目录下有/gi.或者说怎样将/gi目录关联进rviz软件。

freemagicfan, 2020-02-07:
# Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version of GAAS
- [ ] I checked the Gitbook documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I Googled/Binged the error messages generated by Ubuntu, Gazebo, and PX4
- [ ] I have posted questions regarding Ubuntu, Gazebo, and PX4 in corresponding forums.

# Issue Template

## Context
Please provide any relevant information about your setup.
* Tutorial No.:[eg. Tutorial 4]
* Mavros: [eg. 0.18.4]
* ROS: [eg. Kinetic]
* Gazebo: [eg. 7]
* Ubuntu: [eg. 16.04]
* System Platform: [eg. X86]

## Expected Behavior

Please describe the behavior you are expecting.

## Current Behavior

What is the current behavior?

## Screenshots
If applicable, add screenshots to help explain your problem.

## Failure Logs

Copy console log and paste it here.

MasterPa, 2020-02-07:
Please use English to ask the question. Thank you!

MasterPa, 2020-02-07:
And please follow the intro to ask, which could let us help you better.

hddgi, 2020-02-10:
The "/gi/navi_path_plan" is a ROS topic not a directory, you can see more detail about rostopic at [ROS Wiki](http://wiki.ros.org/ROS/Tutorials). 

RVIZ is a 3d visualization tool for ROS applications. It provides a view of your robot model, capture sensor information from robot sensors, and replay captured data. The "/gi/navi_path_plan" is a ROS topic which publish the "visualization_msgs/Marker" messages, and you can visualize the Marker Display data in rviz.

closed 2020-02-17 by hddgi