generalized-intelligence/GAAS #89: Point Cloud not show up in Glvz

nba2kolqw, 2020-02-15:
I am working on the tutorials E04. However point cloud is not showing up. I think it is because ROS is not publishing. And it will abort when I run "sh run.sh" Can someone help me with this issues please? 
[image]
[image]
[image]

hddgi, 2020-02-17:
`/cloud_in` will be published correctly if `sh run.sh` is run successfully. But `sh run.sh` runs wrong on your computer. Please make sure your repo is up-to-date, and maybe compile it again.

MasterPa, 2020-02-24:
Close the issue, reopen if needed.

closed 2020-02-24 by MasterPa