generalized-intelligence/GAAS #93: EO4 生成相机坐标系的环境点云报错

hp-cuiwb, 2020-02-26:
# Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- I am running the latest version of GAAS
- I checked the Gitbook documentation and found no answer
-  I checked to make sure that this issue has not already been filed

## Context

your gitbook ubuntu mirror

## Screenshots
[image]

## Failure Logs
root@cwb-System-Product-Name:~/GAAS/software/Obstacle_Map# sh run.sh 
-------------Brief-----------------
left_img_topic: /gi/simulation/left/image_raw
right_img_topic: /gi/simulation/right/image_raw
calib_file_name: calibration/mynteye2.yml
calib_width: 752
calib_height: 480
out_width: 752
out_height: 480
debug: 0
disparityMax: 20
disparityMin: 2
distanceMax: 10
distanceMin: 0.1
conductStereoRectify: 0
RangeOfDisparity: 16
SizeOfBlockWindow: 7
PreFilterSize: 255
PreFilterCap: 63
SmallerBlockSize: 7
MinDisparity: 7
NumDisparities: 32
TextureThreshold: 255
UniquenessRatio: 115
SpeckleWindowSize: 0
Lambda: 0
SigmaColor: 10
displayImage: 0
-----------------------------------
double free or corruption (out)
Aborted (core dumped)
root@cwb-System-Product-Name:~/GAAS/software/Obstacle_Map# ./bin/match_pointcloud_and_slam_pose2
using radius: 0.1
using points number: 100
using mean_k: 50
using stdThres: 1
Node initialized as: pointcloud_to_global_frame
sync_pc_callback 2

hp-cuiwb, 2020-02-27:
能帮助解决下吗

hddgi, 2020-02-27:
I failed to reproduce your problem, Can you determine the error line number by adding log output (etc. use "std::cout") or debugging with gdb? 
The source code of the `sh run.sh` execution file is at `Obstacle_Map/src/dense_reconstruction.cpp`

hp-cuiwb, 2020-02-27:
SpeckleWindowSize: 0
Lambda: 0
SigmaColor: 10
displayImage: 0
-----------------------------------
[New Thread 0x7fffcbbd5700 (LWP 7861)]
[New Thread 0x7fffcb3d4700 (LWP 7862)]
[New Thread 0x7fffcabd3700 (LWP 7863)]
[New Thread 0x7fffca3d2700 (LWP 7864)]
double free or corruption (out)

Thread 1 "dense_reconstru" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: 没有那个文件或目录.
(gdb) thread inf
inf                     inflate@plt             inflateCopy             inflateGetHeader        inflateMark             inflateReset@plt        inflateSyncPoint        infosev
inf_zero                inflateBack             inflateCopy@plt         inflateInit2_           inflatePrime            inflateResetKeep        inflateUndermine        
infinity                inflateBackEnd          inflateEnd              inflateInit2_@plt       inflateReset            inflateSetDictionary    inflateValidate         
infint                  inflateBackInit_        inflateEnd@plt          inflateInit_            inflateReset2           inflateSync             inflateValidate@plt     
inflate                 inflateCodesUsed        inflateGetDictionary    inflateInit_@plt        inflateReset2@plt       inflateSync@plt         info(char const*, ...)  
(gdb) thread inf
inf                     inflate@plt             inflateCopy             inflateGetHeader        inflateMark             inflateReset@plt        inflateSyncPoint        infosev
inf_zero                inflateBack             inflateCopy@plt         inflateInit2_           inflatePrime            inflateResetKeep        inflateUndermine        
infinity                inflateBackEnd          inflateEnd              inflateInit2_@plt       inflateReset            inflateSetDictionary    inflateValidate         
infint                  inflateBackInit_        inflateEnd@plt          inflateInit_            inflateReset2           inflateSync             inflateValidate@plt     
inflate                 inflateCodesUsed        inflateGetDictionary    inflateInit_@plt        inflateReset2@plt       inflateSync@plt         info(char const*, ...)  
(gdb) thread inf
inf                     inflate@plt             inflateCopy             inflateGetHeader        inflateMark             inflateReset@plt        inflateSyncPoint        infosev
inf_zero                inflateBack             inflateCopy@plt         inflateInit2_           inflatePrime            inflateResetKeep        inflateUndermine        
infinity                inflateBackEnd          inflateEnd              inflateInit2_@plt       inflateReset            inflateSetDictionary    inflateValidate         
infint                  inflateBackInit_        inflateEnd@plt          inflateInit_            inflateReset2           inflateSync             inflateValidate@plt     
inflate                 inflateCodesUsed        inflateGetDictionary    inflateInit_@plt        inflateReset2@plt       inflateSync@plt         info(char const*, ...)  
(gdb) threads
Undefined command: "threads".  Try "help".
(gdb) threads info
Undefined command: "threads".  Try "help".
(gdb) threads inf
Undefined command: "threads".  Try "help".
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff0cb6801 in __GI_abort () at abort.c:79
#2  0x00007ffff0cff897 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff0e2cb9a "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff0d0690a in malloc_printerr (str=str@entry=0x7ffff0e2e870 "double free or corruption (out)") at malloc.c:5350
#4  0x00007ffff0d0de75 in _int_free (have_lock=0, p=0x555555d25250, av=0x7ffff1061c40 <main_arena>) at malloc.c:4278
#5  __GI___libc_free (mem=0x555555d25260) at malloc.c:3124
#6  0x00007ffff1ae1f93 in cvReleaseMat () from /usr/local/lib/libopencv_core.so.3.4
#7  0x00007ffff1ccfb61 in cv::read(cv::FileNode const&, cv::Mat&, cv::Mat const&) () from /usr/local/lib/libopencv_core.so.3.4
#8  0x000055555558413c in ?? ()
#9  0x0000000042ff0000 in ?? ()
#10 0x0000000000000000 in ?? ()
(gdb) thread
[Current thread is 1 (Thread 0x7ffff7f57280 (LWP 7817))]
(gdb)

Wenjian-Hao, 2020-04-07:
You get this error because ''GAAS/software/Obstacle_Map/src/dense_reconstruction.cpp'' does not get the file "GAAS/software/Obstacle_Map/calibration/mynteye2.yml" which is a matrices file.
You can solve it by 
1. figure out why it does not get the config file(we refer this matrix file in config.yaml/calib_file_name:...), then rerun the run.sh
2. you can define the matrices directly in ''GAAS/software/Obstacle_Map/src/dense_reconstruction.cpp'' then rerun the run.sh, i adopted this and it works

cyanine-gi, 2021-04-27:
> You get this error because ''GAAS/software/Obstacle_Map/src/dense_reconstruction.cpp'' does not get the file "GAAS/software/Obstacle_Map/calibration/mynteye2.yml" which is a matrices file.
> You can solve it by
> Figuring out why it does not get the config file(we refer this matrix file in config.yaml/calib_file_name:...), then rerun the run.sh
> OR you can define the matrices directly in ''GAAS/software/Obstacle_Map/src/dense_reconstruction.cpp'' then rerun the run.sh, i adopted this and it works
Thank you!

closed 2021-04-27 by cyanine-gi