generalized-intelligence/GAAS #94: Tutorial 4 problem running run.sh from Obstacle_Map

danydownload, 2020-02-27:
## Context
Please provide any relevant information about your setup.
* Tutorial No.: Tutorial 4
* Mavros: 1.0.0
* ROS: [Melodic
* Gazebo: 9
* Ubuntu: 18.04
* System Platform: X64

## Expected Behavior
 ruh.sh should generate the disparity map

## Current Behavior
When i run run.sh i have this error "double free or corruption (out)" and the program aborts.
I found that the problem is in the file "(GAAS_PATH)/software/Obstacle_Map/src/dense_reconstruction.cpp"
when he tries to read K1 from yml file but i don't know how to fix it.

## Screenshots
[image]

## Failure Logs

dani@dani-GL502VT:~/catkin_ws/src/GAAS/software/Obstacle_Map$ sh run.sh 
-------------Brief-----------------
left_img_topic: /gi/simulation/left/image_raw
right_img_topic: /gi/simulation/right/image_raw
calib_file_name: calibration/mynteye2.yml
calib_width: 752
calib_height: 480
out_width: 752
out_height: 480
debug: 0
disparityMax: 20
disparityMin: 2
distanceMax: 10
distanceMin: 0.1
conductStereoRectify: 0
RangeOfDisparity: 16
SizeOfBlockWindow: 7
PreFilterSize: 255
PreFilterCap: 63
SmallerBlockSize: 7
MinDisparity: 7
NumDisparities: 32
TextureThreshold: 255
UniquenessRatio: 115
SpeckleWindowSize: 0
Lambda: 0
SigmaColor: 10
displayImage: 0
-----------------------------------
double free or corruption (out)
Aborted (core dumped)

danydownload, 2020-03-04:
I think the problem is that ROS is compiled with opencv 3.2 while the camera application wants opencv 3.4.5.
However i found a workaroud. Just setting calibration parameters from code instead of reading from file.
[image]

An example:
[image]

MasterPa, 2020-03-04:
Thank you! We are also trying to figure out the problem right now:P

closed 2020-04-01 by MasterPa