generalized-intelligence/GAAS #98: [BUG]

Lndulge666, 2020-04-02:
* ROS:  Kinetic
* Gazebo: 9
* Ubuntu:  16.04

在Tutorial 6中，运行set_init.py时出现错误：ImportError: No module named ros_kcf.srv，请问该如何解决

hddgi, 2020-04-08:
Please read our tutorial carefully。 In order to keep the environmental variables lean, you have to manually modify `set_init.py` to add dependencies for ROS Service to the list of environmental variables.

closed 2020-04-10 by Lndulge666

Ryan18281613101, 2020-04-10:
the error still exists after I modified the file set_init.py,line 8,as sys.path.append('/home/GAAS/demo/tutorial_6/GAAS-Object-Tracking-master/KCF/build/devel/lib/python2.7/dist-packages').If I modified in a wrong way?

liangpu597, 2020-04-27:
@Ryan18281613101 I change the file path, and it works, you can try again

omargazebo, 2020-07-07:
@hddgi I did the same steps in the tuto still got the same error: ImportError: No module named ros_kcf.srv . Is there a tutorial that shows more details please ? 
@liangpu597 can you please share the path that you used to make it work.

Myocat, 2024-11-21:
> the error still exists after I modified the file set_init.py,line 8,as sys.path.append('/home/GAAS/demo/tutorial_6/GAAS-Object-Tracking-master/KCF/build/devel/lib/python2.7/dist-packages').If I modified in a wrong way?

You got the file path wrong. There is a username between /home and /GAAS. Like this: /home/(username)/GAAS...