generalized-intelligence/GAAS #99: eo3  执行sh generate.sh出错 出现G2O_SOLVER_CHOLMOD     linked by target "ygz-util" in directory /home/champossion/soft/GAAS/software/SLAM/ygz_slam_ros/util

champossion-svg, 2020-04-11:
eo3  执行sh generate.sh出错 出现2O_SOLVER_CHOLMOD
    linked by target "ygz-util" in directory /home/champossion/soft/GAAS/software/SLAM/ygz_slam_ros/util
    linked by target "ygz-common" in directory /home/champossion/soft/GAAS/software/SLAM/ygz_slam_ros/common
    linked by target "ygz-cv" in directory /home/champossion/soft/GAAS/software/SLAM/ygz_slam_ros/cv
    linked by target "ygz-backend" in directory /home/champossion/soft/GAAS/software/SLAM/ygz_slam_ros/backend
    linked by target "MakeSceneEurocStereoVIO_ros" in directory /home/champossion/soft/GAAS/software/SLAM/ygz_slam_ros/examples
    linked by target "EurocStereo" in directory /home/champossion/soft/GAAS/software/SLAM/ygz_slam_ros/examples
    linked by target "EurocStereoVIO_ros" in directory /home/champossion/soft/GAAS/software/SLAM/ygz_slam_ros/examples
    linked by target "Kitti" in directory /home/champossion/soft/GAAS/software/SLAM/ygz_slam_ros/examples

-- Configuring incomplete, errors occurred!
See also "/home/champossion/soft/GAAS/software/SLAM/ygz_slam_ros/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.

bys1123, 2020-04-11:
这是我在github上见过最长的issue标题了

hddgi, 2020-04-13:
Please confirm that g2o is installed correctly(e.g. have you executed `make install`), and have you install the dependencies correctly mentioned in the [GAAS tutorial 3].(https://gaas.gitbook.io/guide/software-realization-build-your-own-autonomous-drone/build-your-own-autonomous-drone-part-3-using-slam-in-gps-denied-environment-for-position-estimation)

hddgi, 2020-04-13:
Please confirm that g2o is installed correctly(e.g. have you executed `make install`), and have you install the dependencies correctly mentioned in the [GAAS tutorial 3](https://gaas.gitbook.io/guide/software-realization-build-your-own-autonomous-drone/build-your-own-autonomous-drone-part-3-using-slam-in-gps-denied-environment-for-position-estimation).

closed 2020-04-21 by champossion-svg