generalized-intelligence/GAAS #111: Tutorial 03: recipe for maketarget failed

RitvikN, 2020-07-20:
Hello,

I am following Tutorial 3 for SLAM in a GPS denied environment. I have installed all required dependencies, but when I get to `sh generate.sh ` inside GAAS/software/SLAM/ygz_slam_ros I get the following error after 32% installation:

`CMakeFiles/Makefile2:2917: recipe for target 'common/CMakeFiles/ygz-common.dir/all' failed
make[1]: *** [common/CMakeFiles/ygz-common.dir/all] Error 2
CMakeFiles/Makefile2:3027: recipe for target 'backend/CMakeFiles/ygz-backend.dir/all' failed
make[1]: *** [backend/CMakeFiles/ygz-backend.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2`

How do I fix this?

Doodle1106, 2020-07-20:
I tried a clean installation on my machine and I did not see any compiling error. The error message you provided is not enough for spotting the issue so I cannot help you here.

However you can check:
1. is your opencv installed properly?
2. is g2o installed properly?
3. go to Thirdparty/DBow3, delete build and rebuild DBow3 by:
```
rm -r build
mkdir build
cd build
cmake ..
make
```

Doodle1106, 2020-07-20:
You'll need to copy and paste the actual error message so I can help you.

RitvikN, 2020-07-20:
I just did a clean install of Ubuntu 18 and followed all the steps in Tutorial 1 and Tutorial 3 to get the dependencies installed and working. I used the GAAS pre-built dependencies for PCL, DBoW3, Pangolin, g2o and opencv.  Once again, I get an error in Tutorial 3 at `sh generate.sh`. It runs for a few minutes and they stops with makefile recipe failed error. Here is an image of my terminal when the error occurred:

[image]

RitvikN, 2020-07-20:
Re compiled all required packages for SLAM using cmake/make the way you specified above. I also installed opencv 4.3.0 with opencv_contrib modules as specified in the tutorial. All succeeded without any errors. Trying to run `sh ./generate.sh` again resulted in this error:

[image]

Also, when installing opencv through cmake-gui there is no option called ENABLE_CXX11. I had to manually add it as a boolean value. Did not encounter any errors during generation and cmake, so not sure if it made any difference. Here's a picture of the cmake-gui to show you that it's missing:

[image]

Doodle1106, 2020-07-21:
Mavros is missing from the system, since you are on Ubuntu 18.04, try:
```
sudo apt install ros-melodic-mavros-*
```

RitvikN, 2020-07-21:
Ended up doing a fresh install of Ubuntu 18. For some reason, installing PCL, DBoW3 and g2o the way you specified above does not work. When I reach `sh generate.sh` it always fails because it's unable to locate files from those packages. I always have to install the pre compiled binaries provided on the GAAS site for it to work. Do you have any idea why this happens? Here are some pictures of the errors I get:

[image]
[image]

Anyways, I was finally able to complete run `sh generate.sh` without any errors. However, when I proceed to `sh run.sh` to begin running SLAM, I get this error: `Segmentation fault (core dumped)`

[image]

How do I fix this?

Wo-Ki, 2021-07-01:
> Ended up doing a fresh install of Ubuntu 18. For some reason, installing PCL, DBoW3 and g2o the way you specified above does not work. When I reach `sh generate.sh` it always fails because it's [...]

Hi, I have the save question:"Segmentation fault" when I run "sh run.sh". This is about ygz slam, but there is no question when   
 I try the original [ygz-stereo-inertial](https://github.com/gaoxiang12/ygz-stereo-inertial) code on the same Nvidia Jetson Xavier. I hope I can get help,tks.