generalized-intelligence/GAAS #117: No module named pyquaternion

atieh-sahraei, 2020-08-03:
# Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version of GAAS
- [ ] I checked the Gitbook documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I Googled/Binged the error messages generated by Ubuntu, Gazebo, and PX4
- [ ] I have posted questions regarding Ubuntu, Gazebo, and PX4 in corresponding forums.

# Issue Template

## Context
Please provide any relevant information about your setup.
* Tutorial No.:[eg. Tutorial 4]
* Mavros: [eg. 0.18.4]
* ROS: [eg. Kinetic]
* Gazebo: [eg. 7]
* Ubuntu: [eg. 16.04]
* System Platform: [eg. X86]

## Expected Behavior

I followed the installation in Tutorial#1 and it seems that my installation is set for python 2. I want to run the python files to control the drone.

## Current Behavior

But when I run " python px4_mavros_run.py" it gives the following error:

from pyquaternion import Quaternion
ImportError: No module named pyquaternion

I tried to install it with : " pip install pyquaternion ", and the terminal gives:

Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pyquaternion in /home/srl/.local/lib/python3.6/site-packages (0.9.5)
Requirement already satisfied: numpy in /usr/local/lib/python3.6/dist-packages (from pyquaternion) (1.17.4)
WARNING: You are using pip version 20.1.1; however, version 20.2 is available.
You should consider upgrading via the '/usr/bin/python3 -m pip install --upgrade pip' command.

It seems it is installed for python 3.
So, I tried to run the script with "python3 px4_mavros_run.py" 
Do you think that I have to set my environment with python3 ?

## Failure Logs

Px4 Controller Initialized!
Waiting for initialization.
Waiting for initialization.
Waiting for initialization.
Waiting for initialization.
Waiting for initialization.
Waiting for initialization.
Waiting for initialization.
Waiting for initialization.
Waiting for initialization.
Waiting for initialization.
Traceback (most recent call last):
  File "/home/srl/catkin_ws/devel_isolated/mavros_msgs/lib/python2.7/dist-packages/mavros_msgs/msg/_PositionTarget.py", line 168, in serialize
    buff.write(_get_struct_BH9d2f().pack(_x.coordinate_frame, _x.type_mask, _x.position.x, _x.position.y, _x.position.z, _x.velocity.x, _x.velocity.y, _x.velocity.z, _x.acceleration_or_force.x, _x.acceleration_or_force.y, _x.acceleration_or_force.z, _x.yaw, _x.yaw_rate))
struct.error: required argument is not a float

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 882, in publish
    self.impl.publish(data)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 1066, in publish
    serialize_message(b, self.seq, message)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/msg.py", line 152, in serialize_message
    msg.serialize(b)
  File "/home/srl/catkin_ws/devel_isolated/mavros_msgs/lib/python2.7/dist-packages/mavros_msgs/msg/_PositionTarget.py", line 169, in serialize
    except struct.error as se: self._check_types(struct.error("%s: '%s' when writing '%s'" % (type(se), str(se), str(locals().get('_x', self)))))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genpy/message.py", line 364, in _check_types
    raise SerializationError(str(exc))
genpy.message.SerializationError: <class 'struct.error'>: 'required argument is not a float' when writing 'header: 
  seq: 1
  stamp: 
    secs: 255
    nsecs: 250000000
  frame_id: ''
coordinate_frame: 9
type_mask: 1016
position: 
  x: 0
  y: 0
  z: 3.2
velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
acceleration_or_force: 
  x: 0.0
  y: 0.0
  z: 0.0
yaw: None
yaw_rate: 1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "px4_mavros_run.py", line 308, in <module>
    con.start()
  File "px4_mavros_run.py", line 77, in start
    self.local_target_pub.publish(self.cur_target_pose)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 886, in publish
    raise ROSSerializationException(str(e))
rospy.exceptions.ROSSerializationException: <class 'struct.error'>: 'required argument is not a float' when writing 'header: 
  seq: 1
  stamp: 
    secs: 255
    nsecs: 250000000
  frame_id: ''
coordinate_frame: 9
type_mask: 1016
position: 
  x: 0
  y: 0
  z: 3.2
velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
acceleration_or_force: 
  x: 0.0
  y: 0.0
  z: 0.0
yaw: None
yaw_rate: 1'

lucasfoll, 2020-08-19:
Same error here. Did you fix it?

cyanine-gi, 2020-11-07:
Just install the pyquaternion via pip2 or pip3 (which depends the version of your python.), and restart your simulation.

closed 2020-11-07 by cyanine-gi