generalized-intelligence/GAAS #119: can‘t receive image using rviz

StancyMiller, 2020-08-10:
I'm using virtual machine ubuntu 16.04.
When I tried the Basic Object Tracking demo(Tutorial 6), I could open the simulation scene in Gazebo and run init_drone.py.
But when I open rviz and chose “Add”->“by topic”->”/gi/simulation/left/image_raw“ to see the drone's view, it didn't work and showed "no view".
So I wonder if the demo could be run on virtual machine.
[image]

StancyMiller, 2020-08-20:
Ohhhh, it was because of the version of Gazebo. The image can be shown after I update Gazebo to 7.16. The issue could be closed.

cyanine-gi, 2020-11-07:
Maybe you need some patience when using gazebo :)

closed 2020-11-07 by cyanine-gi