generalized-intelligence/GAAS #121: Tutorial 6：Lose the sight of the ground vehicle after executing " track_and_move.py"

CusUdont, 2020-10-15:
Hello,

After I run track_and_move.py, the drone will move backward and its camera will lose the sight of the ground vehicle. After that, the "Track Rect" will stay constant even though the ground vehicle shows up in the sight again. And the drone will hover in the air and cannot complete the tracking mission.

Does anyone know how to solve this? Thanks for the help.

frogotterman, 2020-11-13:
Hello,
I'm facing the same problem. 
Have you been able to solve it?

cyanine-gi, 2021-05-11:
Check your mavros coordinate. It seems mavros use different coordinate in different versions.
Maybe you need to add a coordinate transformation operation in your mavros controller.