generalized-intelligence/GAAS #125: takoff failed

cillian-bao, 2021-01-04:
when I run:rostopic  echo /mavros/state, I got this output:
---
header: 
  seq: 602
  stamp: 
    secs: 602
    nsecs: 312000000
  frame_id: ''
connected: True
armed: False
guided: True
manual_input: False
mode: "AUTO.LAND"
system_status: 3
---
in the end,  Iwhe  I run:python px4_mavros_run.py,I got:
Px4 Controller Initialized!
Vehicle Took Off Failed!
I think it is the mode:"AUTO.LAND"'s problem

malfarizit, 2021-01-06:
hi @cillian-bao 
i think u should change mode to "MANUAL" 

`rosrun mavros mavsys mode -c MANUAL`

or u can restart your roslaunch.

cyanine-gi, 2021-04-27:
Check out your px4 configuration in QGC and set mode to position manually before takeoff may help.
If possible, connect your RC controller and make sure that QGC spot it, so that the drone won't just drop to fail-safe condition.

closed 2021-04-27 by cyanine-gi