generalized-intelligence/GAAS #134: AirSim Tutorial?

NickPerezCarletonUniversity, 2021-07-08:
Hi, I saw someone from your team posted links to an AirSim tutorial on reddit: https://www.reddit.com/r/radiocontrol/comments/cw29bw/tutorial_using_airsim_to_simulate_aircraft/

However, the links are all broken. Do you still have these tutorials available somewhere?

cyanine-gi, 2021-07-18:
All the deprecated codes are place at GAAS/deprecated/ directory.
For your requirement, check: https://github.com/generalized-intelligence/GAAS/blob/master/deprecated/demo/case_study_1/readme.md
Our gitbook is still available now: check https://gaas.gitbook.io/guide/
But it will soon be replaced with the documents of new lidar-based GAAS. The old version of GAAS, which is vision-algorithms based, is not recommended anymore.
GAAS will be a fully lidar-based framework towards L5 autonomous driving flying cars and cargo drones.

closed 2021-07-18 by cyanine-gi

NickPerezCarletonUniversity, 2021-07-19:
Thanks! What made you choose to stop using vision and use lidar instead? Here is a resource showing the most advanced vision and lidar odometry algorithms: http://www.cvlibs.net/datasets/kitti/eval_odometry.php
The most advanced solutions use both vision and lidar combined. I think vision is important to always have, it gives certain information lidar can never acquire.

cyanine-gi, 2021-09-06:
> Thanks! What made you choose to stop using vision and use lidar instead? Here is a resource showing the most advanced vision and lidar odometry algorithms: [...]

Please pay attention that all online SLAM algorithms are not robust enough for flying cars & large cargo drones. So the pre-built map for a scene is required.

Later we may release a vision-based version of GAAS (maybe GAAS-lite or something like this), still using pre-built map, which will be constructed by cameras (Structure from Motion), rather than by Lidars, as a low-cost solution.

NickPerezCarletonUniversity, 2021-09-06:
ah ok. slightly unrelated, but:

Do you know of a good resource for learning about 'active SLAM'? As in, navigating the drone based off of recognizing obstacles from the mapping part of the SLAM algorithm and the SLAM algorithm's localization estimate. I have been learning about SLAM assuming the robot is being controlled or that it's trajectory is already determined.... I want to navigate the drone in AirSim based off the SLAM's position estimation, and navigate around the environment staying as far away from the observed obstacles as possible. For example, if the drone was instructed to fly down a corridor, how would I use the map estimation and localization estimation from a SLAM algorithm to allow the drone to autonomously fly through the **middle** of the corridor (stays as far away from both sides of the corridor's walls as possible).

NickPerezCarletonUniversity, 2021-09-06:
also @cyanine-gi I might recommend also experimenting with something like UWB beacons to help localize the drone.