generalized-intelligence/GAAS #137: roslaunch error. Resource not found: xacro

Lee-Siyoung, 2021-08-25:
## Context
Please provide any relevant information about your setup.
* Tutorial No.:[eg. Tutorial 6]
* Mavros: [eg. 0.18.4]
* ROS: [eg. melodic]
* Gazebo: [eg. 7]
* Ubuntu: [eg. 18.04]
* System Platform: [eg. X86]

I was doing tutorial six. To verify GAAS-Object-Tracking by completing make-j4, roslaunch px4 car_tracking.launch was performed. But it didn't work out as an error.

```
roslaunch px4 car_tracking.launch
... logging to /home/drone/.ros/log/916aaad6-0576-11ec-b4c4-d8c0a6b1244f/roslaunch-drone-desktop-2023.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Resource not found: xacro
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/drone/image_transport_ws/src
ROS path [2]=/opt/ros/melodic/share
ROS path [3]=/home/drone/catkin_ws/src/Firmware
ROS path [4]=/home/drone/catkin_ws/src/Firmware/Tools/sitl_gazebo
The traceback for the exception was written to the log file
```

Below is an error log.
```
[roslaunch][INFO] 2021-08-25 16:32:15,030: Added node of type [gazebo_ros/gzserver] in namespace [/]
[roslaunch][INFO] 2021-08-25 16:32:15,032: Added node of type [gazebo_ros/gzclient] in namespace [/]
[roslaunch][ERROR] 2021-08-25 16:32:15,033: Resource not found: xacro
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/drone/image_transport_ws/src
ROS path [2]=/opt/ros/melodic/share
ROS path [3]=/home/drone/catkin_ws/src/Firmware
ROS path [4]=/home/drone/catkin_ws/src/Firmware/Tools/sitl_gazebo
[roslaunch][ERROR] 2021-08-25 16:32:15,033: The traceback for the exception was written to the log file
[roslaunch][ERROR] 2021-08-25 16:32:15,036: Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 332, in main
    p.start()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/parent.py", line 289, in start
    self._start_infrastructure()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/parent.py", line 238, in _start_infrastructure
    self._load_config()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/parent.py", line 144, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/config.py", line 461, in load_config_default
    loader.load(f, config, argv=args, verbose=verbose)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 761, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 733, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 679, in _recurse_load
    self._param_tag(tag, context, ros_config, verbose=verbose)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 96, in call
    return f(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 263, in _param_tag
    vals = self.opt_attrs(tag, context, ('value', 'textfile', 'binfile', 'command'))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 208, in opt_attrs
    return [self.resolve_args(tag_value(tag,a), context) for a in attrs]
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 189, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 392, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 405, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 166, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 212, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 207, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: xacro
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/drone/image_transport_ws/src
ROS path [2]=/opt/ros/melodic/share
ROS path [3]=/home/drone/catkin_ws/src/Firmware
ROS path [4]=/home/drone/catkin_ws/src/Firmware/Tools/sitl_gazebo

[rospy.core][INFO] 2021-08-25 16:32:15,036: signal_shutdown [atexit]
```

my ~/.bashrc
```
source /opt/ros/melodic/setup.bash
source ~/catkin_ws/devel/setup.bash
source ~/image_transport_ws/devel/setup.bash
source ~/catkin_ws/src/Firmware/Tools/setup_gazebo.bash ~/catkin_ws/src/Firmware/ ~/catkin_ws/src/Firmware/build/posix_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/catkin_ws/src/Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/catkin_ws/src/Firmware/Tools/sitl_gazebo
export GAZEBO_MODEL_PATH=:/home/drone/catkin_ws/src/Firmware//Tools/sitl_gazebo/models:~/GAAS/deprecated/simulator/models
```

closed 2021-08-25 by Lee-Siyoung

wing0716, 2021-08-25:
Hello, I also encountered a similar problem, have you solved the problem?

Lee-Siyoung, 2021-08-25:
> Hello, I also encountered a similar problem, have you solved the problem?

yes. This is how I solved it.
sudo apt-get install ros-<version>-xacro  
<version> is Write according to the ros version
I hope it helps. :)

wing0716, 2021-08-25:
Thanks. It's working now. :)