generalized-intelligence/GAAS #152: Website Down?

hendursaga, 2022-11-25:
[https://www.gaas.dev/]() appears to be down, with a Wix.com page that says this domain isn't connected to a website.

TabLand, 2023-06-30:
It does appear that this project is in hiatus / abandoned?

TabLand, 2023-06-30:
I found the following links:

Gitbook: https://gaas.gitbook.io/guide/ (Last update: 9th March 2020)
Slack: https://join.slack.com/t/gaasdev/shared_invite/zt-8jl4kgyc-vzH_Yl_gS3Q_EonxZjFMgQ (Last admin post: 27 March 2020)
Youtube: https://www.youtube.com/@gaas6888 (Last post: 27 March 2020)
Twitter: https://twitter.com/GAAS_dev (Last original post: 26 July 2021)

Looks like the company was heavily impacted by COVID?

Inglam, 2023-08-12:
Any recent updates? Anyone know any of the memebers on LinkedIn?

TabLand, 2023-08-15:
Hi Inglam,

I have tried to make contact with the company via various channels but not received a reply yet. I think it's safe to assume this project is currently dormant.

This project could be revived under a hard fork and existing PRs could be reviewed and possibly merged. However, without the relevant industry expertise, relevant hardware and more handover information from the original founders - it does feel like a tough challenge.