generalized-intelligence/GAAS #154: double free or corruption (!prev) when roslaunch registration_localization ndt.launch

wangjiac1, 2024-06-02:
# Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version of GAAS
- [ ] I checked the Gitbook documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I Googled/Binged the error messages generated by Ubuntu, Gazebo, and PX4
- [ ] I have posted questions regarding Ubuntu, Gazebo, and PX4 in corresponding forums.

# Issue Template

## Context
Please provide any relevant information about your setup.
* ROS: [eg. Noetic]
* Gazebo: [eg. 11.13.0]
* Ubuntu: [eg. 20.04]
* System Platform: [eg. X86]

## Expected Behavior

## Current Behavior

I run ./build_all.sh and build all successful,but when I run  roslaunch registration_localization ndt.launch,it occurs an error "double free or corruption (!prev) .

## Screenshots
[image]

## Failure Logs

Copy console log and paste it here.

cray168, 2024-06-05:
请问问题解决了么，我这边./build不成功是怎么回事呢

wangjiac1, 2024-06-05:
> 请问问题解决了么，我这边./build不成功是怎么回事呢

报啥错了

cray168, 2024-06-05:
[image]

cray168, 2024-06-05:
CMake Error at LocalizationAndMapping/registration_localization/CMakeLists.txt:104 (add_executable):
  add_executable cannot create target "map_publisher" because another target
  with the same name already exists.  The existing target is an executable
  created in source directory
  "/home/yangzining/111/GAAS/algorithms/src/LocalizationAndMapping/lidar_localization".
  See documentation for policy CMP0002 for more details.

CMake Error at LocalizationAndMapping/registration_localization/CMakeLists.txt:105 (target_link_libraries):
  Attempt to add link library "opencv_calib3d" to target "map_publisher"
  which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.

CMake Error at LocalizationAndMapping/registration_localization/CMakeLists.txt:108 (add_library):
  add_library cannot create target "fast_gicp" because another target with
  the same name already exists.  The existing target is a shared library
  created in source directory
  "/home/yangzining/111/GAAS/algorithms/src/LocalizationAndMapping/icp_lidar_localization".
  See documentation for policy CMP0002 for more details.

CMake Error at LocalizationAndMapping/registration_localization/CMakeLists.txt:114 (target_link_libraries):
  Attempt to add link library "pcl_common" to target "fast_gicp" which is not
  built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.

CMake Error at /usr/local/share/cmake-3.23/Modules/FindCUDA.cmake:1985 (add_library):
  add_library cannot create target "fast_vgicp_cuda" because another target
  with the same name already exists.  The existing target is a shared library
  created in source directory
  "/home/yangzining/111/GAAS/algorithms/src/LocalizationAndMapping/icp_lidar_localization".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  LocalizationAndMapping/registration_localization/CMakeLists.txt:130 (cuda_add_library)

CMake Error at LocalizationAndMapping/registration_localization/CMakeLists.txt:150 (target_link_libraries):
  Attempt to add link library "/opt/ros/noetic/lib/libcv_bridge.so" to target
  "fast_vgicp_cuda" which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.

CMake Error at /usr/local/share/cmake-3.23/Modules/FindCUDA.cmake:2147 (target_link_libraries):
  Attempt to add link library "/usr/local/cuda/lib64/libcublas.so" to target
  "fast_vgicp_cuda" which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.
Call Stack (most recent call first):
  LocalizationAndMapping/registration_localization/CMakeLists.txt:153 (cuda_add_cublas_to_target)

wangjiac1, 2024-06-06:
> CMake Error at LocalizationAndMapping/registration_localization/CMakeLists.txt:104 (add_executable): add_executable cannot create target "map_publisher" because another target with the same name [...]

定位的那三个包可执行文件的名字重复了，把名字改一下或者只编译其中一个就可以了